xieyaxiongfly/Atheros-CSI-Tool-UserSpace-APP #3: cannot receive csi data.

coutinho09, 2016-02-13:
I use a laptop as DP.
OS: Ubuntu 14.10
NIC: Atheros AR9285
After Installation of the Linux Kernel, I check the kernel version, and it returns "4.1.10+".

I use a router as AP with no password.
the Wireless Net parameter:
    channel: auto
    mode: 11n only (or 11bgn mixed)
    bandwidth: auto

When DP ping 192.168.1.1, it returns properly, but  recieve application Atheros-CSI-Tool-UserSpace-APP/recvCSI/recv_csi cannot work properly,  there‘s no data received, the function read_csi_buf() in main.c always returns 0. I changed Wireless Net parameter, such as channel-1 to 10, mode-11n only or 11bgn mixed, bandwidth-20MHz or 40MHz, but it also doesn't work properly.

Thank you!

xieyaxiongfly, 2016-02-13:
Hi there,

I think your problem is caused by the version Wi-Fi NIC you use. Some users has tell me that it is highly possible that Atheros 11n Chipsets with version AR92xx doesn't support CSI extraction. I think your experience confirmed that. 

Thus, my suggestion is to change the Wi-Fi NIC to AR93xx, AR94xx or AR95xx. Detailed  version of Wi-Fi NIC our tool can support can be found on our website.

If it still doesn't work after changing the NIC, please let me know.

Yaxiong

coutinho09, 2016-02-13:
Thanks, I'll try it.

coutinho09, 2016-02-22:
I have changed a new NIC, AR9565, but it also can't receive CSI data. How can I debug it to find where the error is.
Thank you!

xieyaxiongfly, 2016-02-22:
Can you be more specific on your problem? 

What do you mean by cannot receive CSI data? How did you set up your environment? What's the output? Please give a detailed description about your experiment and the output or debug information so that I can help to find out the cause.

Yaxiong

xieyaxiongfly, 2016-02-22:
I read your first post and find one problem.

Did you use two laptop that installed our TOOL to conduct the experiment? Our tool requires the sender and receiver to be a computer that has our tool installed.

bboyzhou, 2016-07-22:
I installed CSI tool in two PCs,and I use three 5G directional antennas at TXD
NIC:AR9380
OS:Ubuntu 14.10

I encounter some problem when I use the CSI tool.

The receiving terminal can receive the packets, but when I process the data,I found some (even all) csi matrixs in the csi trace are empty.  No matter the  transmitting antenna is opposite the antenna or not, the problem always happen.  I found some abnormal datas in csi trace when the csi matrix is empty :
csi_len: 641 ,rate :149 ,nc: 0 and payload_len: 1924 . The corresponding data are 1710, 143, 2, 1040 when the csi matrix isn't empty.  I can't find the reason,so I write to you for help.

Thank you!

xieyaxiongfly, 2016-07-25:
This is strange. I cannot tell the problem from your description. Personally, I didn't meet this problem before. Can you send the received CSI data to me so that I can take a look at it ?

bboyzhou, 2016-07-26:
This is the  received CSI data.

 
         
  

从QQ邮箱发来的超大附件

csi.dat (24.78M, 2016年08月25日 09:30 到期)进入下载页面：http://mail.qq.com/cgi-bin/ftnExs_download?t=exs_ftn_download&k=0a39633161a316986f80441a47335749570b075553515d5f1b5d5505521e5157040f4e500355044b540c510007555255020855016134650545504d550047655b&code=69c1a3ef

EuclidGH, 2017-03-11:
Hello,

I have the same problem as bboyzhou. I also get playload len 1924 which actually destroys the log.dat (csi is empty).
I use a PC and Laptop, both run Ubuntu 14 LTS and bothe have installed  Atheros CSI Tool.

Furthermore, I would always get only these packages, which I sent for example from PC laptop, but It does not work always. Sometimes I get packages with payload len 28, 104, etc, which do not destroy the log.dat.
The problem occurs once the payload len 1924 is. I can not find the source, where from it comes. 

Could somebody help me?

Also, the problem is, when I start the recv_csi and nothing has been still sent to it, it receives packages with different lengths, one of them is 1924 length. 

Where is a problem? 
How to solve it.

Plese, help me. This is very important, becuase I use this tool for my master thesis. 

Thanks

bboyzhou, 2017-03-11:
Thought I haven't found the reason of that problem, I solved it by add a conditional statement (if(length(csi_trace{ii}.csi)==1)  continue; )in my Matlab code(csi_trace = My_read_log_file(file);),  because when csi_len<=0,csi_matrix.csi = 0 in the read_log_file.m.  I hope it helps you.

EuclidGH, 2017-03-11:
Thanks for your answer.
SO, do not get payload len 1924 any more or you get it still, but filter it?
could you send me you matlab code? 
k-ieso@hotmail.com
thanks

EuclidGH, 2017-03-12:
Hello again,

Sorry for disturbing, but I wanted to ask you about the shit payload len 1924, which destroys the csi data.
You told me that you added a conditional statement and it helped you to fix the problem. Your conditional statement does not change anything for the protocol, where CSI data will be transmitted and then extracted. It means you should get the payload len 1924 in any cases. Furthermore, I have one more problem with little and big endian. Normally, before the packages will be sent, the sendData.c checks the system, whether it is big or little endian. After that it sends the packages to the destination. The recv_csi writes the csi with already included endian. So, when I get payload len 1924, the matlab prints error that there is a wrong endian. I think it means, this shit 1924 payloads do not come from the transmitter. In another case, there could be a right endian. 

P.S I send 50 packages always, because when I send one package, I don’t know what happens between receiver and transmitter that the packages are lost, or they are received, but not written in log.dat. That’s why I send 50 packages.

Also, xieyaxiongfly, please comment 
Best wishes

mzakharo, 2022-04-18:
Please see https://github.com/xieyaxiongfly/Atheros-CSI-Tool-UserSpace-APP/issues/22#issuecomment-552143069  on whats causing 1924.