xieyaxiongfly/Atheros-CSI-Tool-UserSpace-APP #4: CSI with Ad-hoc networks

BrunoSoares-LABORA, 2016-04-01:
Hi,

It's possible to receive CSI data with ad-hoc networks?

Att.,

xieyaxiongfly, 2016-04-03:
Hi,

Actually, my answer is I don't know also.

I am glad to say that I don't know since our tool has so many possible applications that I am not able try all of them. 

Our tool is totally opensource and all of its functionality are implemented in software. Therefore, you can try it all by yourself and you can add whatever functionality to the tool as you like. 

I strongly recommend you to try it yourself :)

Yaxiong

BrunoSoares-LABORA, 2016-04-08:
Hi,

I'm asking that because i have tested the tool with a ad-hoc network but i can't receive CSI data by the "recvCSI" application. I think i have do something wrong because now i'm tried using your hostapd modified application to make the CSI computer a master AP and after that i connected my smartphone and my notebook to this network and executed a ping command to the AP and my result is no one packet in the recvCSI application.

The kernel of the AP computer where the CSI tool was installed is OK in version "4.1.10+" and i have executed the installation according your tutorial.

Can you help me?

Best Regards.,

BrunoSoares-LABORA, 2016-05-05:
Problem solved.

closed 2016-05-05 by BrunoSoares-LABORA