xieyaxiongfly/Atheros-CSI-Tool-UserSpace-APP #5: which version of Atheros X NIC work properly?

xiaolingu, 2016-04-08:
Hi 
According to your knowledge about users' response to your CSI-tool in email and so on, Which version of Atheros NIC can work successfully in practice ? 
Looking forward to your reply !

Nanjing
@xieyaxiongfly

xieyaxiongfly, 2016-04-08:
Hi,

Till now, the following version has been proven working successful. 

AR9590
AR9580
AR9565
AR9462
AR9380

Thanks

xieyaxiongfly, 2016-04-08:
And you can find the information at there
http://pdcc.ntu.edu.sg/wands/Atheros/DeviceList.html

xiaolingu, 2016-04-15:
Thank you for your rapid response！！
In addition，I have another question whether I need to identify the BIOS in my laptop？My laptop likely
set a access for hardware equipment.How do you deal with the problem ?

xieyaxiongfly, 2016-04-27:
Yes, you need to modify the BIOS in the laptop to bypass the hardware blacklist.

Sorry for answering late, coz I was traveling these days.

xiaolingu, 2016-05-14:
Excuse me ， I have installed the CSI tool in a PC . However when using the read_csi file to read the specific file , CSI item doesn't include any data and CSI length == 0 .Should I use another computer including Atheros NIC as a Soft AP ? 

Thanks

xieyaxiongfly, 2016-05-17:
Yes. You must have another PC installed with our tool and configured as AP.