xieyaxiongfly/Atheros-CSI-Tool-UserSpace-APP #6: sounding flag

libing64, 2016-05-01:
@xieyaxiongfly 
Great thanks for sharing the excellent tools. 
Can you recommend some papers or books talking about the sounding flag especially the packet format correlated to the flag?

libing64, 2016-05-01:
@xieyaxiongfly 
I want to add injection/monitor mode to current csi tool.
My idea is to use a modified version of [aireplay-ng](https://github.com/aircrack-ng/aircrack-ng/blob/master/src/aireplay-ng.c) to inject a frame with souding flag turned on, then the receiver can collect csi data without connectimg to the AP. 
However I am not familiar with communication and now I am confused by kinds of frame formats in ieee80211n standard. Can you provide some help to construct the mac frame with sounding flag turned on? Great thanks.

xieyaxiongfly, 2016-05-06:
Hi there,

I can tell you how did I do to turn on the sounding flag in my tool.

Under the ath9k driver folder, you can find a file named "ar9003_mac.c". In that file, you can find some code in the following figure.
[image]
You can refer to this figure to understand how I turn on the sounding flag.

And I didn't touch or I cannot touch the PHY layer header of one packet. The PHY layer header is added at the MCU of the wifi card. I cannot change that part. But the good thing is that, the driver will tell the wifi card what kind of PHY layer header to be constructed. Therefore, I change the message that the driver send to the card and tell the card to construct a PHY layer header with sounding flag turning on.

Hope that will help.

libing64, 2016-05-06:
@xieyaxiongfly 
Thanks for your kind reply. 
Maybe the PHY layer header is added in the drivers code. You can see this [source file](https://github.com/aircrack-ng/aircrack-ng/blob/master/src/aireplay-ng.c).
I tried the aircrack-ng tools, in fact, it could generate any packet to send. 
For example, here is the authentication packet generated in aireplay-ng. 
#define AUTH_REQ        \
    "\xB0\x00\x3A\x01\xBB\xBB\xBB\xBB\xBB\xBB\xCC\xCC\xCC\xCC\xCC\xCC" \
    "\xBB\xBB\xBB\xBB\xBB\xBB\xB0\x00\x00\x00\x01\x00\x00\x00"

What I want to know is the packet format for the sounding frame. After knowing the packet format, the frame can be injected and sent to the receiver directly, without requirement for the ath9k driver.
By setting the receiver monitor mode and the transmitter injection mode, the receiver could collect csi date from multiple transmitters at the same time. 
I think many people have talked about the possibility on Atheros NIC, especially some localization guys.  

Since I am not familiar  with communication, please help me.

xieyaxiongfly, 2016-05-06:
I think you are talking about the MAC header, is that right? Modifying the MAC header is very simple. But the PHY header is quite difficult to access and hard to modify. 

I am not familiar with the aircrack-ng tool. Are you sure it can modify the PHY layer header?

libing64, 2016-05-06:
@xieyaxiongfly  
I double checked aircrack-ng, it only modifies the MAC header, not the PHY header. 
Sorry.

xieyaxiongfly, 2016-05-06:
No problem at all. 

But I don't get why you need to do it by yourself? I already changed the sounding flag in my tool.

libing64, 2016-05-07:
@xieyaxiongfly 
I know you have turned on the sounding flag in the ath9k driver, and I have successfully received the csi  data with your tools. Thanks for your excellent work.
But the problem is that it can only receive csi from one transmitter, but what I want is multiple transmitters. That why I need the injection/monitor mode. 
The injection/monitor mode works well on Intel 5300 NIC card to collect csi, but now I am think about the possibility on Atheros NIC.

xieyaxiongfly, 2016-05-07:
Maybe you can try this injection mode directly. I am pretty sure that sounding flag has nothing to do with the injection mode. 

I think what you need to do is using injection program to replace the sendData program in our tool.

libing64, 2016-05-09:
@xieyaxiongfly 
Thanks, I will have a try.

closed 2016-05-09 by libing64

xieyaxiongfly, 2016-05-31:
@libing64 
We can support multiple receivers now. Please refer to our websites for more information.

libing64, 2016-05-31:
@xieyaxiongfly Wow, cool.

navidtadayon, 2017-06-12:
@xieyaxiongfly 
Hi,
I sent you a message a while ago and asked you a question about this CSI tool. Thank you for doing the superb  job of developing this tool and making it available to the community. I have been struggling for long with the same problem that @libing64 mentioned above: I am able to collect CSI in master mode. What I need to do is to be able to collect CSI when the receiver is in monitor mode. The reason I insist on this is that, in monitor mode, steering matrices from more number of channels can be collected within a given amount of time. However, as soon as I put the receiver in this mode, nothing is collected. When I inspected your modified driver  code, I realized that rx_not_sounding=1, rx_hw_upload_data=0, rx_hw_upload_data_valid=0, rx_hw_upload_data_type=0. These values obviously block the driver to collect anything. In fact, they might mean that there is nothing to collect! I do not understand why this is happening since (according to my way of working out the theory) ath9k requires steering (CSI) matrix for decoding HT packets in monitor mode, so the latter will have to be calculated by the driver no matter in which mode (what do you think?). On the other hand, above-mentioned parameters are calculated based on 11 words of RX descriptor that is made available by the driver in ar9003_mac.c. This makes me believe that, perhaps, Atheros cards do calculate steering matrices in monitor mode but do not upload it on the hardware. Long story short, I am seeking your advice! In particular, I read your paper "precise power delay profiling using commodity wifi" and it seems you were able to do the same thing that I need to do... Any feedback is highly appreciated.
Navid