xieyaxiongfly/Atheros-CSI-Tool-UserSpace-APP #11: Varying dimensions of CSI matrices while moving the receiver

ClarenceZSK, 2016-10-24:
Hi Dr. Xie,

When I am moving the receiver, the CSI matrix (nr*nc*num_tones) will vary in dimension. The sender attached two antennas and the receiver attached three antennas. Most of the received CSI matrices are 3*2*56. However, some are 3*1*56. It also happens when I used the Intel 5300 CSI tool. It seems to depend on the relative positions between antennas of sender and receiver.

Could you please help me explain that? I have no idea why this happens. 

Thank you for your brilliant tool. Look forward to your reply.

Shengkai Zhang
Wuhan, HUST

xieyaxiongfly, 2016-10-25:
I think it happens because of the antenna selection.

At a different location, the system will automatically select different antenna configuration to best deliver the data. If it selects to transmit using 2 antennas, the dimension will be different from transmission using 1 antenna. 

The solution is to fix the transmission configuration. I will find a way to fix the data rate in Ubuntu and release a guide on how to do that.

Yaxiong

jiuyuesmile, 2017-06-12:
Hello @xieyaxiongfly 

Have you found the way to fix the transmission configuration?

Guanzhong

mzakharocsc, 2017-06-18:
```
sudo ifconfig wlan0 down
sudo iw phy phy0 set antenna 0x3
iw phy phy0 info | grep -i ant
sudo ifconfig wlan0 up
```

xieyaxiongfly, 2017-07-10:
To fix the data rate, you can use a simple command:

echo 0 >> /sys/kernel/debug/ieee80211/phy#/rc/fixed_rate_idx

The index  0 means the data rate is 0x80. You can change it to 1 and the rate becomes 0x81.

shulinyang, 2019-04-18:
请为您解决这个问题了吗？因为我也遇到同样的问题？您是如何修复Ubuntu中的数据速率呢？期待您的回答@ClarenceZSK