xieyaxiongfly/Atheros-CSI-Tool-UserSpace-APP #12: CSI for openwrt?

wynter-wang, 2016-11-04:
Hello!
    I would like to know if you have the experience of running CSI-Tool on openwrt system?
    Looking forward your reply!
   Thanks a lot!

xieyaxiongfly, 2016-11-08:
Yes, we do. We have already done that. We are preparing for the release.

wynter-wang, 2016-11-09:
Thanks for your replay! Recently I have been ported csi-tool to openwrt system, the version of my openwrt system is 14.07, the Linux kernel is 3.10.49, and the chip is ar9580. Unfortunately, I met some questions in the process, in the file of ar9003_csi.c, I grabbed some data, but the result of rx_hw_upload_data and rx_hw_upload_data_type are always be zero, which result in csi->pkt_status.csi_len = 0, and the csi->csi_buf is null. So, I wish if you could spare some time to explain the reason for this situation, I will be very grateful.

xieyaxiongfly, 2016-11-09:
I cannot tell the reason. It could be caused by a lot of possible reasons.

Can you explain more?