xieyaxiongfly/Atheros-CSI-Tool-UserSpace-APP #13: anyone can tell me how to get the subcarrier frequence ?

linkzhao, 2016-11-08:


xieyaxiongfly, 2016-11-08:
First of all, check the channel you are transmitting on, for example, 2.435GHz, which is the central frequency of your channel.

Then, I believe, the central frequency can be somehow viewed as the frequency of the central subcarrier. 

With the frequency interval (312.5khz) between two subcarriers, you can infer the frequency of other subcarriers of your channel.

linkzhao, 2016-11-08:
Thanks,I got it !

EuclidGH, 2017-01-23:
Hi,

could you tell how did you get to run the my_log.dat in matlab?

I need the help... Thank you

chenwq3, 2017-03-30:
You need to re-compiled your recvCSI and sendData.

522623905, 2017-04-15:
Hi.anyone could tell me the zero-subcarrier fre is equal to the center frequency ?