xieyaxiongfly/Atheros-CSI-Tool-UserSpace-APP #16: How to Process log file in MATLAB

asifhanif1122, 2017-01-18:
Hi,
We have successfully logged file with CSI and now want to process it in MATLAB. Can you please guide how to extract CSI from received file using MATLAB scripts or any other option that is available. 
We previously used Intel 5300 CSI tool in which we used to obtain dat file and later extract CSI from file using MATLAB scripts provided by  the author of CSI tool ,Daniel Halperi.

I have read your user guide but could not get the desired help. It would be very helpful if you provide a little bit guidance about processing of log file obtained using your Atheros tool. Thanks!

Regards,
Asif Hanif 
asif.hanif@lums.edu.pk

xieyaxiongfly, 2017-02-10:
Hi there,

We have a matlab folder in our UserSpace applications. In that folder, you can find a matlab function for processing the CSI just like intel 5300 tool.

Yaxiong

chenwq3, 2017-03-28:
Hi,

I'm new to MATLAB and I am learning how to achieve indoor positioning.

I logged the csi information from `recv_sci` too. And I want to process it in MATLAB too.
I search *intel 5300 tool* and hope to find how to use this tool, but for now I find nothing.

```bash
$ ls 
csi.dat  read_csi.c  read_csi.mexa64  read_log_file.m  sample_bigEndian.dat
```

```matlab
>> read_log_file sample_bigEndian.dat  % I know this will "create" the array.
```

So how do I do to convert csi.data to another_bigEndian.dat??
Do you have the url that link to the manual of intel 5300 tool?
Thanks.

Best Regards,
Weiqi Chen.

dindokk, 2017-12-26:
hi xieyaxiongfly where is the folder