xieyaxiongfly/Atheros-CSI-Tool-UserSpace-APP #19: Please help to find solution

EuclidGH, 2017-02-09:
First of all, I would like to explain our scenario of the measurement setup.
1.	We want to use the tool as ping-pong and then we to do analyzing of CSI data. It means that after execution of command sudo ./send_data wlan0 MAC_OF_RECV 10 we get 10 or less than 10 packets on the receiver, which is shown (we suppose) via Recv X-th msg…., where X means the number of received packets, we want to send the answer back to the transmitter. This is like a pinging of an IP, where you get a response of each ping. It seems that your tool does not support such function. To achieve our purpose, we used the sendData.c and modified it and now after printing Recv X-th msg… is called the function sendData(total_msg_cnt) which sends a new packet back to the transmitter. However, this is very bad, because we would have the same packet back, which we sent to the receiver. Could you please let us know, how we can get the same packet back on the transmitter?

2.	The next modifying was that we used the function process_csi  to print the CSI status in the terminal, where we can see the whole information about CSI data (this saves a lot of time because we do not need to use MatLab to see the CSI status while testing). Sometimes we get 0 CSI-Timestamp and sometimes we get the value 1092516660480 microseconds which is always the same and never changes. Also, we get sometimes 0 or 2 for the number of transmitting antenna, although we use 3x3 antennas (AR9380). This is also strange. Could you suppose something that can have an influence on it? 

3.	One more problem is with the AP. We could not get an AP under 802.11n standard. We want to use 5 GHz for our measurement setup. The last way was to configure the hospapd under the 802.11a standard, which supports 5 GHz. However, this standard does not support a MIMO technology, which is very important to obtain a CSI data. During the configuration of 802.11n standard, we got always the error that the device does not support the hw_mode. We tried to use an another Atheros NIC (AR9380), but is has nothing changed. The same problem with AP. (all devices has been chosen according to the list of supported devices. Could you also tell something about this problem?

Please help to fine a solution..... 

Best wishes,
Ieso

xieyaxiongfly, 2017-02-10:
For the first question, you cannot simply call the sendData function. The sendData function always send exactly the same packet. If you want to achieve the goal, you need to change it. You can look at this part of sendData.c 
 
This part construct the packet. As you can see, the payload of the packet is 1000 0xaa. You can change this part and replace the 1000 0xaa with whatever you want to transmit. 

For the second question, I don’t know what’s the reason for the timestamp. I just record whatever the Wi-Fi card told me. About the antenna, it means that the transmitter will change its transmission configuration packet by packet. Last packet, it may use two antenna and next packet it may decide to use 3. If you get 0 for tx antenna, it doesn’t make any sense. I don’t know what’s the reason and I never met that before.

For constructing an AP in 5G, I recommend you use a Wi-Fi router with openWRT installed. I am now preparing a tutorial about how to install our tool on top of Wi-Fi router.

EuclidGH, 2017-02-10:
Thank you very much for your reply.

mzakharocsc, 2017-06-17:
for #2, there seems to be an integer overflow in the code, just recompile with -Wall, and fix the compiler warnings. However, I am not sure how to relate the values in this field to system/epoch time.

master-li, 2017-11-30:
For the second question, I found it because of the author's careless extraction of the timestamp in the wrong format. I modified it after extracting timestamp verify that it is correct. That place is the record_status function under the csi_fun.cwe file.

[image]

  | for example |  

(buf_addr[0] << 56) & 0x00000000000000ff) | ((buf_addr[1] << 48) & 0x000000000000ff00)

modify in
(buf_addr[0] << 56) & 0xff00000000000000) | ((buf_addr[1] << 48) & 0x00ff000000000000)

@xieyaxiongfly 
@EuclidGH

xieyaxiongfly, 2017-12-01:
Get it.

junstriker, 2019-04-03:
Hello,
I' having an issue with the timestamp. I corrected the csi_fun.c file, lines 141-144 and 152-55 (as per http://lion.sjtu.edu.cn/project/projectDetail?id=5) and recompiled. 
However I'm not sure if the timestamp is correct when I extract the csi data on Matlab, for example, in one packet I get 3.1670e9 microseconds and when i convert that to normal date/time I get Friday, February 6, 1970 3:43:20 PM  which is obviously not my current date/time.
I'm not sure if I'm interpreting the timestamp correctly. The date/time on my linux machine is updated. Can you tell me what I'm doing wrong, or how to I get the correct timestamp in the collected csi data?

Many thanks.

mzakharocsc, 2019-04-03:
This timestamp is radio timestamp, not  linux date/time. Not sure how they are related. If you need linux date/time, replace the value you get in the packet with the return of gettimeofday().

junstriker, 2019-04-03:
@mzakharocsc  
Thank you very much for this clarification. I'm just getting started with all the CSI extraction and analysis concept. I wanted to know what information the radio timestamp basically provides. For example if I have one packet with a timestamp of 2.2445e9 microseconds and the next one with a timestamp of say 2.2465e9 (payload length=1024 bytes), what can I infer from that? How this information is important in analysis, especially in wifi passive sensing? Furthermore, I can plot the CSI amplitude with respect to number of packets or subcarriers per packet. How to I plot the amplitude with respect to time?

Many thanks again.
Kind regards