xieyaxiongfly/Atheros-CSI-Tool-UserSpace-APP #23: OpenWRT, csi-len = 0

chaosmaster, 2017-03-23:
Hi there,

I am trying to get this working on a **TP-LINK TL-WA901ND v4** Access Point using OpenWRT.

I have sucessfully ported the modified driver from here:
https://github.com/xieyaxiongfly/Atheros-CSI-Tool
to OpenWRT, but was having issues with it.

I have then found
https://github.com/xieyaxiongfly/AtherosCSI_openWRT_version
And am still having the same issues.

I have attached patches for both versions.

According to the openwrt-wiki the AP uses the **AR9380** chipset:
https://wiki.openwrt.org/toh/tp-link/tl-wa901nd

Dmesg however reports AR9561:
[    9.533407] ieee80211 phy0: Atheros **AR9561** Rev:0 mem=0xb8100000, irq=47

When using either patch I am allways getting csi_len = 0.

(As you can see I have added it to the output of recv_csi)

Any suggestions, what could be going wrong?

(Both receiver and transmitter use the same hardware with the patch)

Thank you.

> 
> root@OpenWrt:~# send_Data wlan0 60:E3:27:4D:C9:63 1000
> Packet Length is: 1014,pkt_num is: 1000

> root@OpenWrt:~# recv_csi 
> /**************************************/
> /*   Usage: recv_csi <output_file>    */
> /**************************************/
> #Receiving data! Press Ctrl+C to quit!
> Recv 1th msg with rate: 0x96 | payload len: 92 | csi len: 0
> Recv 2th msg with rate: 0x96 | payload len: 1040 | csi len: 0
> Recv 3th msg with rate: 0x96 | payload len: 1040 | csi len: 0
> Recv 4th msg with rate: 0x96 | payload len: 1040 | csi len: 0
> Recv 5th msg with rate: 0x96 | payload len: 1040 | csi len: 0
> Recv 6th msg with rate: 0x96 | payload len: 1040 | csi len: 0
> Recv 7th msg with rate: 0x96 | payload len: 1040 | csi len: 0
> Recv 8th msg with rate: 0x96 | payload len: 1040 | csi len: 0
> Recv 9th msg with rate: 0x96 | payload len: 1040 | csi len: 0
> ...

[560-ath9k_csi_tool.patch.txt](https://github.com/xieyaxiongfly/Atheros-CSI-Tool-UserSpace-APP/files/865036/560-ath9k_csi_tool.patch.txt)
[ath9k-openwrt-new.patch.txt](https://github.com/xieyaxiongfly/Atheros-CSI-Tool-UserSpace-APP/files/865034/ath9k-openwrt-new.patch.txt)

EuclidGH, 2017-03-23:
what does your patches exectly?

chaosmaster, 2017-03-23:
@EuclidGH, the patches just implement the modifications to the ath9k-driver.
They can just be placed into the openwrt-buildroot (package/kernel/mac80211/patches/) so the OpenWRT build-system can patch ath9k directly.

EuclidGH, 2017-03-23:
ok.
sorry, I can not help you. I use just two devices (pc, and laptop) for my experiments.
Please check the last post here https://github.com/xieyaxiongfly/Atheros-CSI-Tool-UserSpace-APP/issues/22

maybe you will help me to find a solution.

niklasser, 2017-03-23:
Hey,
i got it working with an AR9344 and chaos calmer.

What i did:
Build OpenWrt from https://github.com/RedSnake64/openwrt-yun without any Packages.
Applied CSI patch to mac80211 as  **LAST** patch.
Install kernel package from /bin/base/packages to openwrt

chaosmaster, 2017-03-23:
@niklasser which patch did you use?
I am using openwrt trunk and added my patch as 560-ath9k...
whis is the last patch for ath9k.

niklasser, 2017-03-24:
@chaosmaster we uploaded our Patches [here](https://github.com/NovelSense/ath9k_csi_kernel).
You can find a litte tutorial there. 
I hope it helps.

chaosmaster, 2017-03-29:
@niklasser, thanks for your patch.
Now with your patch it seems I am unable to receive anything when sending from one (patched) AP to another (patched) AP, it just sits there waiting for data:
```
root@OpenWrt:~# recv_csi
/**************************************/
/*   Usage: recv_csi <output_file>    */
/**************************************/
#Receiving data! Press Ctrl+C to quit!

```

```
root@OpenWrt:~# send_Data wlan0 60:E3:27:4D:C9:62 1000
Packet Length is: 1014,pkt_num is: 1000

```

Using different (unpatched) Hardware for sending packets i am still getting csi_len = 0:

`Recv 12th msg with rate: 0x8f | payload len: 1040 | csi len: 0`

Any ideas?

EuclidGH, 2017-03-29:
How do you print the csi_len variable?

it is u_int16_t type and you need to import inttypes.h and thenprint as following
printf("%" PRIu16 "\n", csi_len);

check it.

chaosmaster, 2017-03-29:
I am using "%d", as is done with payload_len which is also defined as uint16_t:

`printf("Recv %dth msg with rate: 0x%02x | payload len: %d | csi len: %d\n",total_msg_cnt,csi_status->rate,csi_status->payload_len,csi_status->csi_len);`

I don't think the output is the issue here, the logfile doesn't contain any CSI either.

scholz, 2017-03-29:
Hi chaosmaster,
this may be trivial, but have you tried to turn around the roles of AP and client i.e. run recv_csi on AP and send_csi on client and vice versa?

niklasser, 2017-03-29:
Hey @chaosmaster,
you wrote 

> from one (patched) AP to another (patched) AP

.
Are you actually using both devices as AP? If so, you should use one as AP and the other as client. In my measurements i sent packets from the client to the AP.

By the way, you should be able to find the exact Atheros-Chip by entering uboot. It would be interesting to know what chip you are actually using.

chaosmaster, 2017-03-29:
@scholz thanks for the input, I did try it both ways.
@niklasser I have one AP configured as AP and the other connected as a client.
The connection itself is fine, I can ping one from the other.
I will check out what uboot says, thanks for the tip.

xieyaxiongfly, 2017-04-05:
@chaosmaster 

You can try the bin file I have compiled. I didn't have WA701 so I didn't personally test it. I only test it with TL-WDR4300.

xieyaxiongfly, 2017-04-05:
@chaosmaster 
 I just upload the bin to a new git repository. You can check that.

chaosmaster, 2017-04-05:
@xieyaxiongfly Thanks for your reply, and for uploading the bin.
It seems using the bin from the new repo (openwrt-ar71xx-generic-tl-wa901nd-v4-squashfs-sysupgrade.bin) I am hstill having the same issues.
Sending data from an unmodified laptop I am getting:`
Recv 12th msg with rate: 0x8f | payload len: 1040 | csi len: 0`

Using the firmware on two identical APs and one configured as a client connected to the other I am not getting anything with **recv_csi** although ping works just fine (albeit having pretty high latency):

```
root@OpenWrt:~# ping 192.168.1.183
PING 192.168.1.183 (192.168.1.183): 56 data bytes
64 bytes from 192.168.1.183: seq=0 ttl=64 time=60.241 ms
64 bytes from 192.168.1.183: seq=1 ttl=64 time=59.900 ms
64 bytes from 192.168.1.183: seq=2 ttl=64 time=59.911 ms
64 bytes from 192.168.1.183: seq=3 ttl=64 time=60.045 ms
```

xieyaxiongfly, 2017-04-05:
That's strange. 
Can you try this two bin file for send data and receive data and let me know the output?

SendDataLoop: https://www.dropbox.com/s/z2ykfk1woivadtl/send_Data_loop?dl=0
RecvCSI:            https://www.dropbox.com/s/q25hmoflfl7zaci/recvCSI?dl=0

chaosmaster, 2017-04-06:
I can't seem to get anything with one AP as client, I am also wondering what could cause the latency to be that high:
```
root@OpenWrt:~# ping 192.168.1.1
PING 192.168.1.1 (192.168.1.1): 56 data bytes
64 bytes from 192.168.1.1: seq=0 ttl=64 time=60.546 ms
64 bytes from 192.168.1.1: seq=1 ttl=64 time=60.304 ms
64 bytes from 192.168.1.1: seq=2 ttl=64 time=59.901 ms

```

Here some link info:
```
root@OpenWrt:~# iw wlan0 link
Connected to 60:e3:27:4d:c8:e4 (on wlan0)
        SSID: OpenWrt
        freq: 2462
        RX: 1469760 bytes (21449 packets)
        TX: 312239 bytes (6251 packets)
        signal: -34 dBm
        tx bitrate: 130.0 MBit/s MCS 15

        bss flags:      short-preamble short-slot-time
        dtim period:    2
        beacon int:     100

```

When using an unmodified Laptop connected to the AP ping looks fine:
```
ping 192.168.1.1
PING 192.168.1.1 (192.168.1.1) 56(84) bytes of data.
64 bytes from 192.168.1.1: icmp_seq=1 ttl=64 time=24.5 ms
64 bytes from 192.168.1.1: icmp_seq=2 ttl=64 time=0.387 ms
64 bytes from 192.168.1.1: icmp_seq=3 ttl=64 time=0.365 ms
```

This is the output with the unmodified laptop as sender:
```
Recv 944th msg with rate: 0x86 | payload len: 1040
========  Bsic Information of the Transmission ==========
tstamp= 2145625192  |payload_len= 1040  |csi_len= 0    |channel= 2462
buf_len= 1065    |phyerr= 0          |noise= 0        |rate= 86
chanBW= 1        |num_tones= 114     |nr= 3           |nc= 1
rssi= 170         |rssi_0= 59         |rssi_1= 52      |rssi_2= 58
Recv 945th msg with rate: 0x86 | payload len: 1040
========  Bsic Information of the Transmission ==========
tstamp= 2145625192  |payload_len= 1040  |csi_len= 0    |channel= 2462
buf_len= 1065    |phyerr= 0          |noise= 0        |rate= 86
chanBW= 1        |num_tones= 114     |nr= 3           |nc= 1
rssi= 170         |rssi_0= 59         |rssi_1= 52      |rssi_2= 58
Recv 946th msg with rate: 0x86 | payload len: 1040
========  Bsic Information of the Transmission ==========
tstamp= 2145625192  |payload_len= 1040  |csi_len= 0    |channel= 2462
buf_len= 1065    |phyerr= 0          |noise= 0        |rate= 86
chanBW= 1        |num_tones= 114     |nr= 3           |nc= 1
rssi= 170         |rssi_0= 59         |rssi_1= 52      |rssi_2= 58
Recv 947th msg with rate: 0x85 | payload len: 1040
========  Bsic Information of the Transmission ==========
tstamp= 2145625192  |payload_len= 1040  |csi_len= 0    |channel= 2462
buf_len= 1065    |phyerr= 0          |noise= 0        |rate= 85
chanBW= 1        |num_tones= 114     |nr= 3           |nc= 1
rssi= 170         |rssi_0= 60         |rssi_1= 52      |rssi_2= 58
Recv 948th msg with rate: 0x84 | payload len: 1040
========  Bsic Information of the Transmission ==========
tstamp= 2145625192  |payload_len= 1040  |csi_len= 0    |channel= 2462
buf_len= 1065    |phyerr= 0          |noise= 0        |rate= 84
chanBW= 1        |num_tones= 114     |nr= 3           |nc= 1
rssi= 170         |rssi_0= 61         |rssi_1= 53      |rssi_2= 59

```

Thanks for looking into this

chaosmaster, 2017-04-11:
I now also have a laptop fitted with an AR9380 module and did some further testing.
When it is connected as a client to the TP-Link AP i can receive CSI on the laptop when i send data from the AP:
```
Recv 474th msg with rate: 0x94 | payload len: 1040 | csi len: 1260
Recv 475th msg with rate: 0x94 | payload len: 139 | csi len: 840
Recv 476th msg with rate: 0x8f | payload len: 1040 | csi len: 840
```

The other way around (using the AP as receiver) I am still not getting anything.

Also the issue with the ping remains with the modified notebook connected to the AP:
```
64 bytes from 192.168.1.1: icmp_seq=1 ttl=64 time=30.7ms
64 bytes from 192.168.1.1: icmp_seq=2 ttl=64 time=30.8ms
64 bytes from 192.168.1.1: icmp_seq=3 ttl=64 time=114ms
```

Any ideas?

wdprez, 2018-02-16:
@chaosmaster 
Did you find a solution in the end? I am trying to send data from a TP-Link router to another TP-Link router (both TL-WR1043ND Ver:4.0) and I get the same problem of having nog CSI data (csi len = 0). Dmesg reports the same chipset as in your case (AR9561), so I might have the same issues as you do.

leewez, 2018-05-16:
@chaosmaster  did you have any progress in this issue. very appricitated if you could sheare us anything

salmanpolito, 2020-01-08:
I am also having the same issue, csi_length=0, (using Atheros C7-V2)

mzakharo, 2022-04-18:
The reason for csi_length=0 is for 3x3x114 CSI payload, the entire CSI cannot fit into one DMA transaction, and gets split up into 2  DMA transactions. Since this is unsupported in the original CSI extraction patches, you get the cryptic csi-len=0 error. One solution is to go to 20Mhz bandwidth, or use less antennas ( 3x2x114 works). If you really need 3x3x114, you need to patch the driver:

Please see the patch below that resolves the issue for me:
[csi_len0.patch](https://github.com/mzakharo/openwrt-csi/blob/csi/csi_len0.patch)