xieyaxiongfly/Atheros-CSI-Tool-UserSpace-APP #25: CSI on 3x2 links looks very strange

avishekfsu, 2017-03-28:
Hi I noticed a very weird issue with the CSI collected in some of my experiments. So I have a 3x3 sender setup and a 3x3 receiver setup. Now when I collect the data, some of the packets have a 3x3x56 structure whereas some have a 3x2x56 structure. This issue has been discussed here before and I believe it's due to the rate selection algorithm . But when I look at the csi for a pakcet with 3x2x56 structure the absolute value of the normalized csi looks like this for a transmitting antenna 

[image]

This is very strange because the absolute value for the csi shouldn't be sinusoidal like this . This problem is not there when the structure is 3x3 as the csi for another packet that is recorded about only 2ms earlier than the one above for the same transmitting antenna.

[image]

The above figure looks fine, as the absolute value is generally a slow changing curve.

Could you help me explain this? Is there anything I'm understanding wrong here ?

For reference, I have added a link to the test file and the packets in question are 28717 and 28725 and the plot is for transmitter 1

https://drive.google.com/open?id=0B4W1uTjVcCFOVnoyUnZINkRxZlE

Thanks
Avishek

EuclidGH, 2017-03-28:
Hi there,

could you give more information about your setup?

which atheros NICs do you use and on which Linux distribution runs your tool?
This is very strange, but more strange is that when I have 3x3x114 or 3x3x56 structure, my payload is 1928 bytes, which let the csi matrix completely empty. For more information, I refer you to my posts here https://github.com/xieyaxiongfly/Atheros-CSI-Tool-UserSpace-APP/issues/22

Did you had the same problem?

I am waiting for an answer from Xie.

avishekfsu, 2017-03-28:
Hi 

I used the AR5BHB112 card which I believe uses the AR9380 chipset. I have 2 laptops with 3 antennas in them.

 I have seen packets with csi_len = 0 where the csi matrix is empty, this just happens randomly. I'm not sure what the problem is. But it doesn't happen too often, and for my purposes I can just discard them. Have you tried fixing the data rate?

Thanks

EuclidGH, 2017-03-29:
Hi, 
Anyway, it has to do with 3 streams. I think the problem occurs while processing 3  streams. Die told that in this case ylI should lower the rate, but i don't have any ideas, how can I lower it. There is only one way. You can find the rate in the code and lower it, but this is bad idea i think.

Did you check my last post?
I found some implementation failures. There are posted under the link from above

EuclidGH, 2017-03-29:
Could you send me your matlab script k-ieso@hotmail.com
thank you very much.

mzakharocsc, 2017-06-18:
We had the same issue in our setup communicating between 2 TL-WDN4800 (AR9380), with 3x3 antenna setup. One thing that helped is limiting the number of antennas to 2x2. Here is how:
```
sudo ifconfig wlan0 down
sudo iw phy phy0 set antenna 0x3
iw phy phy0 info | grep -i ant
sudo ifconfig wlan0 up
```

fchen22, 2017-06-21:
Not all APs support that command. The TL-WDR4310 failed
"command failed: Operation not supported (-122)"

mzakharocsc, 2017-06-21:
yeah - that -122 error is misleading. It says 'not supported', when the interface is busy/in use by something else. Maybe something is holding the interface up. Make sure after ifconfig wlan0 down, the interface completely disappears from your ifconfig dump.

fchen22, 2017-06-29:
Thanks a lot. I will try that tonight and update the final result later

fchen22, 2017-06-30:
Yes, now we can set the number of antennas to 2x2.  Unfortunately ,sometimes we got 2x1. Maybe fixed the MCS index should be better solution?

shulinyang, 2019-04-17:
请问您知道如何修复MCS索引吗？感谢@fchen22