xieyaxiongfly/Atheros-CSI-Tool-UserSpace-APP #26: CSI_dev. What does is?

EuclidGH, 2017-04-05:
Hi there,

I would use two Atheros NICs on one PC. Therefore, I run two identical recv_data and both try to use the same file /dev/CSI_dev. Is it ok, or have I to separate that file that each NIC and recv_data will have the own CSI_dev file? Is it possible?

And what does this file and who writes data into?

Thanks.

xieyaxiongfly, 2017-04-05:
I didn't try that yet. 

I recommend you to give a try and let me know the results. Maybe they will write into the same file and you can obtain the data from both card from that file.

EuclidGH, 2017-04-05:
So, our scenario looks so: We have PC and Laptop (bothe run the same Linux Kernel on Ubuntu 14 LTS and to bore are connected the same model of Atheros NICs). We implement sendData.c into the recvCSI (see recv_csi_b), which sends a packet back after receiving. The scenario looks as following:

                    A (PC)                                                                                B (Laptop)
                    send_data.c 100 pkts ---------------------------------->      recv_csi_b + send_data.c
                                                                                                                                   |
                    recv_csi_a  <--------------------------------------------------------------|

This is our idea, to have a response for each received packet by B. It works properly. However, this setup does not work using only one PC. We want to use this setup using only on PC equipped with two Atheros NICs. When we run the recv_csi_a and b at the same time, both access the CSI_dev file. I think the data will be read from that file and then it will be written into log.dat. But, the problem is, I think both NICs try to write the CSI_dev at the same time. This is just our suppose.

As a result, the sent packets will not be received. In some cases, they are received, but with a big delay.

What does the file CSI_dev and can we separate this one for each NIC?

Here you will find our implementation of your code for a response.
https://www.dropbox.com/s/thlqsza7okak560/csi_a_b.zip?dl=0

Thanks.

xieyaxiongfly, 2017-07-10:
I think you have to merge the send_data.c and recv_csi.c into one program so that you only need to access the CSI_dev one time. You can differentiate the CSI you received by looking at it frequency if you are using two wifi at 2.4/5GHz. That's the better way.