xieyaxiongfly/Atheros-CSI-Tool-UserSpace-APP #29: How to determine the correspondence between the antenna and the matrix？

jiuyuesmile, 2017-06-11:
Hi，
When i used your tool in Openwrt，the CSI matrix looks strange sometimes. Some of them are 3*3*56，and some of them are 3*2*56 . How can I know that which antenna of the sender is not sending the data ?
Thanks
Xu

EuclidGH, 2017-06-21:
Hi,

This is a good question. I would like also to know, how actually the number of receiver antennas will be chosen at the data receive time. I suppose that it will be automatically chosen either by the driver or the 802.11n standard and MIMO technology. In the handbook of this standard, I could not find any information. 

Waiting for an answer for this question.

xieyaxiongfly, 2017-07-10:
Generally, the 3 transmitting/receiving antennas of one Wi-Fi router have some order which is predefined by the router. We can denote them as antenna-1, antenna-2 and antenna-3. To know which antenna is antenna-1 or antenna-2 you can conduct a very simple experiment to figure it out.

The core idea is that: in our CSI reading, we have four RSSI value: RSSI, RSSI-1, RSSI-2, and RSSI-3. The RSSI-1 is the RSSI reading of antenna-1 and the same for antenna-2 and antenna-3.

To conduct the experiment, you just unplug one of your 3 receiving antennas, check the RSSI value and check which RSSI has dropped significantly. If it is RSSI-1, then the antenna you have unplugged is antenna-1. Repeating the process and you will figure out all 3 antennas. The setting is the same for both transmitting antenna and receiving antenna. It is also them same for one type of Wi-Fi router.

After that, I can tell something I have learned from my experience.
1, if you are using 1 antenna for transmitting, you can use any of the 3 antennas to transmit and you will get a 3x1x56 CSI matrix. To know which antenna is transmitting, you can check the setting of your system. Or you can unplug anttena to test and figure out.

2, if you are using 2 antennas for transmitting (just as in your case, you get 3x2x56 CSI), it will always be antenna-1 and antenna-2 that are transmitting. You can also test it by unplug the antennas. You can unplug all the antennas and put it back 1 by 1 to check which antenna make the RSSI increases significantly.

abhinav2938, 2019-02-27:
@chandramahankali Hi, I am also using the same router(TP-LINK TL-1043ND v3.x). Where you able to receive the data in 1(tx) 3(Rx) format? And, how much is your sampling rate because my CSI sample is being received very slowly (1 CSI packets/sec) i.e. 1Hz. Is it the same for you or did you increase the sampling rate? If you could clarify this it would be a great help for me.

chandramahankali, 2019-03-05:
Hi Abhinav, yes, I was able to receive the data 1(tx) X 3 (rx) format. please follow the below link to set the txantenna (on your tansmitter) and rxantenna (on your receiver) options. 
https://openwrt.org/docs/guide-user/network/wifi/basic.
My sampling rate 1000 Hz.

abhinav2938, 2019-04-04:
Hi @chandramahankali , could you tell me how did you set the sampling rate? because I tried increasing the speed of sending data packets by decreasing the usleep value as mentioned [here](https://github.com/xieyaxiongfly/Atheros-CSI-Tool-UserSpace-APP/issues/31), but still the packets are collected slowly at the receiver side and the number of packet doesn't corresponds to that specified in the sendData command.

chandramahankali, 2019-04-04:
Yes, by changing the usleep, you can modify the number samples generated. 
the receiver tool dumps packets to a file. "Writing to file" on the router is slow. So, the circual buffer that exists in the driver which stores the CSI packet often get overwritten if the receiver don't read in time. So, what you could do is, re-direct the packets from the receiver to a high performarmance machine like a PC through a socket (write a socket application, and keep it in the server mode. Modify the receiver application on the router to connect to this application and redirect the CSI data to this application. The socket application on the router can log the  CSI data to a file).

abhinav2938, 2019-04-04:
That seems to make sense, thanks for the information, I will try doing that and could you provide your email so that if I have an issue I could directly ask you.

shulinyang, 2019-06-25:
您好，请问一下，我现在接收端数据一直为空怎么解决呢？ubuntu版本@abhinav2938

muhmaz3, 2019-07-15:
@chandramahankali I am trying to achieve transmission configuration like yours. 1(tx) x 3(rx). Using the link you have mentioned I set txantenna = 1 on the transmitter and rxanteena = 3 on the receiver side but when I send data using sendData I am getting csi in 2 x 2 by 56 format. I am not sure if we need to disable diversity? any suggestions.

zhoubaoz, 2020-07-18:
hi，hava you solve the problem that some of the data are 3X2X56?

chandramahankali, 2020-08-03:
sorry, not working on this anymore.

zhoubaoz, 2020-08-03:
thank you！

| |
电子科技大学_信软学院_周保
|
|
zhoubao_z@163.com
|
签名由网易邮箱大师定制