xieyaxiongfly/Atheros-CSI-Tool-UserSpace-APP #30: Mac address extraction from received packets while monitoring mode

huhu426, 2017-07-18:
Hi,
now, I turned receiver mode from managed to "monitor mode" to receive any packet while receiver do not even connect with transmitter(or any AP)
So now, I can receive various packets including packets that destination of them is not receiver itself. However, I can't extract mac address from packet. I think packets are pre-processed from somewhere and "buf_addr" in "recvCSI/csi_fun.c" is processed variouble not including mac_addr or ip addr of transmitter.

If I'm right, How could I extract mac_address? or How could I distinguish various transmitter?
Thank you.

mzakharocsc, 2017-07-18:
the mac address is inside the payload. You can try dumping the raw bytes and opening them in wireshark to get a better visualization/decoding tool. You have to offset a few bytes into the payload for wireshark 'import binary' to be able to read it. Just manually set wireshark to 802.11 MAC frame decode.

xieyaxiongfly, 2017-07-18:
@mzakharocsc gives the correct answer. Thanks, bro.