xieyaxiongfly/Atheros-CSI-Tool-UserSpace-APP #31: set the speed of send packages

hzka, 2017-07-21:
How to set the speed of send packages,the sending package is too fast,10000 packages only need 4 seconds,i need it slow.Can anyone know?
                                                                                                                  he

xieyaxiongfly, 2017-07-21:
In the sendData.c file, line 170, you will find the following code:

if (usleep(50) == -1){

Just change "50" into  larger value, for example, 2000000 if you like and recompile the code. Done

hzka, 2017-07-22:
Dear Xie,
    I install the tool according to the user guide and thanks very much! However, i have two questions, 
    Fistly，when i process the receievd data off line on matlab, there exists some problems. The detailed problems are as shown in the following figures. When i process the received data on MATLAB in windows system, i first mex the "read_csi.c" and load the data, however, the progarm runs error and it shows out of memory, while i process the received data on MATLAB in Ubuntn system, it also doesn't work. I also process the sample data "sample_bigEndian.dat", and it shows the length is 0. The following figures are the detailed error information. 
  Another question is that how to choose the working channel and the frequency of the hostapd.  We successfully open the hostapd and its working frequency is 2437MHz, however, how to change the working channel, such as channel 6 or channel 11, how to change the working frequency so that it can work on 5GHz? 	We try to open the url "10.10.0.1", however it doesn't work.
   Thanks very much! Looking forward to your early reply.

xieyaxiongfly, 2017-07-22:
I didn't see any figures.

To set the channel, you can use "iw" software of Ubuntu to control he channel. 

I strongly recommend you to use Wi-Fi router and OpenWRT as the sender. You can control all those parameters through Luci.

hzka, 2017-07-25:
ok，Thank you for your reply.the figure problem is done,But the channel setting is not solved.According to your advice,I use "iwconfig wlan0 channel 1",the system displays " Device or resource busy";I use"iwconfig wlan0 mode monitor channel 1",It also displays "Device or resource busy"(the relative figure is in appendix),I can't solve this question,Can you help me,Thank you!

nwuzj, 2017-07-25:
Hi， 
    Thanks for your tool, you said that WiFi-router can be as the sender, however, when we try to use a WiFi router as the sender, the receving rate is very slow and the length is 0.

hzka, 2017-07-25:
The channel(frequency)is not changed by iwconfig.I use command:"ifconfig wlan0 down"、"iwconfig wlan0 mode monitor"、"iwconfig wlan0 freq 5.2G"、"iwconfig wlan0",Now i can see the mode is monitor,the freq is 5.2G,but unfortunately，i use "./start_hostapd.sh" command and it can't open wifi.
  I doubt that the problem is the file in hostapd ,it's channel is static,i should change sth in files.Could you give me some advice???Thank you!

maimogu, 2017-07-27:
@nwuzj I have the same question,  I used the tp_link router as the sender, But it doesn't work.

xieyaxiongfly, 2017-07-28:
The tp link router must be installed with our modified openWRT system. Not
every router works.

YanZhus, 2018-11-19:
> The channel(frequency)is not changed by iwconfig.I use command:"ifconfig wlan0 down"、"iwconfig wlan0 mode monitor"、"iwconfig wlan0 freq 5.2G"、"iwconfig wlan0",Now i can see the mode is monitor,the [...]

hi! have you solved this problem，because i have the same problem now！！

abhinav2938, 2019-02-24:
> The tp link router must be installed with our modified openWRT system. Not every router works.
> […](#)
> -- *-- Yaxiong Xie / **谢亚雄* School of Computer Engineering Nanyang Technological University, Singapore
Hi! I am using your modified OpenWRT system on Tp-link router and am able to collect the CSI data also. But my sample is being collected very slowly. Could you please tell how can I increase the rate of collection of CSI data.

YanZhus, 2019-02-25:
> The tp link router must be installed with our modified openWRT system. Not every router works.
> […](#)
> -- *-- Yaxiong Xie / **谢亚雄* School of Computer Engineering Nanyang Technological University, Singapore

i use ubuntu version,now actually i can change the channel by changing hostapd,but i don't know how to fix HT40,if i fix rate,then it fix HT20,i dont know how to change to HT40

shulinyang, 2019-02-28:
How to change the channel by changing hostapd

shulinyang, 2019-02-28:
hi,how to change the channel by changing hostapd@YanZhus

shulinyang, 2019-02-28:
> > 必须使用我们修改的openWRT系统安装tp链路路由器。不是每个路由器都有效 
> > [...](#) 
> > -  *  - _Xaxiong_ Xie / * _谢亚雄_计算机工程学院，新加坡南洋理工大学
> 
> 我使用ubuntu版本，现在实际上我可以通过更改hostapd来更改频道，但我不知道如何修复HT40，如果我修复速率，那么它修复HT20，我不知道如何更改为HT40

> > The tp link router must be installed with our modified openWRT system. Not every router works.
> > […](#)
> > -- *-- Yaxiong Xie / *_谢亚雄_ School of Computer Engineering Nanyang Technological University, Singapore
> 
> i use ubuntu version,now actually i can change the channel by changing hostapd,but i don't know how to fix HT40,if i fix rate,then it fix HT20,i dont know how to change to HT40

hi,how to change the channel by changing hostapd@YanZhus

YanZhus, 2019-03-13:
> hi,how to change the channel by changing hostapd@YanZhus

change the code of hostapd.conf,you can change the channel set

shulinyang, 2019-03-13:
具体怎么修改hostapd.conf的代码呢，希望得到您的帮助@YanZhus

YanZhus, 2019-03-13:
> 具体怎么修改hostapd.conf的代码呢，希望得到您的帮助@YanZhus

把channel的值改了就好了啊

shulinyang, 2019-04-02:
如何更改为HT40@YanZhus

shulinyang, 2019-04-11:
Ubuntu模式下如何更改工作频率以使其可以在5GHz上工作@hzka