xieyaxiongfly/Atheros-CSI-Tool-UserSpace-APP #32: Receiving probe request/response packet in monitoring mode

huhu426, 2017-07-25:
Hello,

I want to receive probe request/response packets in monitoring mode.
Now, I'm already in monitoring mode and tremendous packets comes in.
However, I couldn't find any probe r/r packets at all.
I both check it with "Wireshark" and Wireshark can capture probe r/r packets.
So, How could I capture both normal data packets and probe request/response packets?
Thank you for your help.

Jaewon Kim

xieyaxiongfly, 2017-10-24:
I have implemented the injection mode on OpenWRT version. It will be released soon.

Yue-M, 2018-11-14:
@xieyaxiongfly 
Hello, I want to use the monitor mode, the transmitter uses the openwrt system router, then the receiver can use the Ubuntu system computer?