xieyaxiongfly/Atheros-CSI-Tool-UserSpace-APP #34: setting up the drivers

junsujang, 2017-09-07:
Hello,

I have installed the kernel version as documented.

My system comprises of a receiver and transmitter both using AR9820 with 1x1 for now (ultimately, I would like to use 1(tx) x 3(rx)) on a X-200 Thinkpad.

I tried calling sendData ifName mac nPackets, but all of the packets compain "send failed" (I have configured the ifName to be wlan0 on rules.d).

I figured that I have not configured the wifi-cards properly (i.e. monitor mode, injection mode, antenna selection, channel frequency, etc...). Do you have an example of how to setup the hardwares so that the packets can be sent with success?

Thank you

xieyaxiongfly, 2017-09-12:
How about the mac address? Have you set it to the destination address? To
set up, you just start an AP and let another device connect to that AP,
then the system will configure the channel automatically.

junsujang, 2017-10-05:
Hello,

I have setup the AP and let the receiver connect to that AP.
I see that the hostpad has connection with the mac address of the receiver.
However, when I call 

./send_Data wlan0 mac 1000

The sender outputs "send failed" and I am not sure how I can debug this.

Also, the receiver seems to be receiving data on its own every 30 seconds or so of payload len 68 even when I am not sending anything. 

Have you experienced this before?

closed 2017-10-08 by junsujang

army-mehak, 2019-04-14:
Hey @junsujang 
After setting up AP on the machine which is connected to the router, how did you connect your other machine to it?

And my receiver shows "failed to open device" error.