xieyaxiongfly/Atheros-CSI-Tool-UserSpace-APP #35: read_csi.c error

junsujang, 2017-10-08:
I was able to get a CSI data using the tool. However, when I try to use MATLAB code provided, it complains as follows. I initially thought that this was an error due to my mac computer, but I ran this again on linux, but same error has occured. Has anyone experienced this before?

Error using read_csi
Requested 4294967299x140673063845944x140675699690528
(17179869184.0GB) array exceeds maximum array size preference.
Creation of arrays greater than this limit may take a long time and
cause MATLAB to become unresponsive. See array size limit or
preference panel for more information.

junsujang, 2017-10-08:
I have figured out the first problem. 

When calling mxCreateNumArrray(ndims, dim, class, complex), dims is of type (size_t *). The array "size" was not of correct type, and I was getting an undefined behavior from MATLAB.

However, as I run it, I still have MATLAB crash due to segmentation fault that I cannot track...

rs8650769, 2017-10-08:
you remove this line from the recvCSI and run 'make'.

https://github.com/xieyaxiongfly/Atheros-CSI-Tool-UserSpace-APP/blob/38a2f47a95ca39ec90794a883e0623a2bf0ac407/recvCSI/main.c#L80

then use the updated binary to receive CSI data.
and try again.

junsujang, 2017-10-08:
Unfortunately, I still face the same issue...

junsujang, 2017-10-09:
The problem was with read_log_file.m. My system was using little endian, and 
`endian_format = 'ieee-be';` needed to be changed to `endian_format = 'ieee-le';`

Also, MATLAB was not able to parse upto proper field_len, so I had to add in some customization.

closed 2017-10-09 by junsujang