xieyaxiongfly/Atheros-CSI-Tool-UserSpace-APP #37: Collect CSI from UDP broadcast

mzakharocsc, 2017-10-13:
Hello, Is it possible to receive CSI from UDP brodacast using this tool?

We have AP and client receive CSI working using ICMP pings. However, if we have client receive UDP broadcast packet coming from AP, we receive data on client via tcpdump, however no CSI is recorded.  

P.S.
[AP code](https://gist.github.com/mzakharocsc/76c91a4bdae3e2a99c5b6f630b377fd3):

Client code:
`tcpdump -i wlan0 port 54545 -XX`

xieyaxiongfly, 2017-10-24:
I think it doesn't work.  

From my experience, the Wi-Fi card won't calculate CSI for packets with the destination address not set to itself. Which means the packet you have broadcast is received but CSI is not calculated.

I have implemented a packet injection mechanism for our tool but I only tested on OpenWRT. 

All clients (in either client mode or monitor mode) are configured with the same MAC address. The Sender injects packets with that MAC address as destination and all clients will calculate CSI.

lqh929289158, 2017-11-24:
on the website 
http://pdcc.ntu.edu.sg/wands/Atheros/ 
You mentioned the Broadcast Mode provided by Kun Qian, but the tutorial link is not available. Would you mind checking it?

jinzhang-unsw, 2018-12-17:
The wiki mentions the broadcast mode which associate with AP and change the MAC address. Is it available for Ubuntu version or only for OpenWRT? I try this approach in Ubuntu but cannot record CSI. In details, I change the MAC address and use iwconfig to associate with AP but no CSI is recorded by receiver. Is there any solution? Or the OpenWRT is the only solution currently?