xieyaxiongfly/Atheros-CSI-Tool-UserSpace-APP #39: Phase jump while the receiver moving

rs8650769, 2017-11-16:
Hello,
These days I have some measurements at a parking lot(a flat area and other WiFi signals is very low), the transmitter stay still, the receiver is moving with a constant speed. 
I plot out the heatmap of phase based on the packets' sequence, most of the time the phase changes smoothly(after linear regression). But some times there is a obvious jump between two packets, and some times later it will jump back. 
[image]
You could have a look this figure, X-axis is packets' sequence, Y-axis is the 114 subcarriers.
I have plotted out the critical points. The following figures are the places where the jump happens:
[image]
[image]
There is another critical point:
[image]
[image]
I could not find the reason why the jump happens. In my opinion, the phase should change smoothly while the receiver moving with a constant speed.
The only phenomenon I have found is that the phase flip at the subcarrier whose fading is great, and the flip point of the phase looks like get an extra Pi offset. 

Have you met this problem while you perform your experiment? Do you know the reason why and when this jump occurs and do you have solutioins to get rid of it?

It confused my several days and I'm waiting for your reply. Thank you in advance. 
Shu

xieyaxiongfly, 2017-12-01:
I haven't conducted such experiments to see the phase value.

But theoretically, it is reasonable to have phase change significantly where the amplitude changes a lot, as shown in your figure. Generally, amplitude changes significantly at some frequencies (subcarriers) means we have experienced deep fading on those frequencies. Accordingly, the phase value will also experience a sudden change in that frequency. To understand why, you can check the group delay of phase. I won't cover the details here.

I think, you have gone to some places with deep frequency selective fading and you will observe such phase variations.

mojtabaaghayi, 2019-06-06:
Help me how i install csi?how i process in matlab?
Help me ?send perfct document
sm.aghayi@gmail.com