xieyaxiongfly/Atheros-CSI-Tool-UserSpace-APP #40: How to send injection packet without associate the access point?

kangqf, 2017-12-19:
Hi xie,
    I find the sendData code must send the data by associating the access point, if not, the data won't come to the driver. Can you help me to find some method to send data in  Not-Associate mode?  I want to set the NIC use the command `iw dev wlan0 interface add mon0 type monitor`  and stop the network-manage just like the intel 5300 packet injection setting. But when I do that and use sendData does not call the ath9k driver. ):

xieyaxiongfly, 2017-12-19:
Our packet injector only supports OpenWRT version. Which version do you use?

kangqf, 2017-12-19:
I use the pc version, what's the difference pc and OpenWRT for also using linux wireless subsystem? 
If the  packet injector code supports OpenWRT released?

xieyaxiongfly, 2017-12-19:
It's the same. All of my work are based on OpenWRT. I just didn't
specifically do that for Ubuntu. It needs some modifications. I will
release the code for Ubuntu when I finished it.

kangqf, 2017-12-19:
That's awesome! But can you give me some ideal about the method to inject packet in monitor mode? It's better if you can give me the OpenWRT version code as reference.

kangqf, 2018-01-02:
Hi, xie. I use the Lorcon library to inject the packet to the NIC, Luckily I get the packet use wireshark. But I find the packets can't trigger the CSI record. Can you help me to solve the problem, May be it can be solve in the Tx side as I am consider.

xieyaxiongfly, 2018-01-03:
Hello, what kind of version are you using? Ubuntu or OpenWRT? Did you check
my Wiki
<https://github.com/xieyaxiongfly/Atheros_CSI_tool_OpenWRT_src/wiki/Collect-CSI#injector-monitor-mode>
here?

kangqf, 2018-01-03:
I use Ubuntu, I find when I use lorcon to inject the packet, the rate that the TX used is below 0x80, so, it doesn't trigger the CSI record. But I am confusing about the reason. Can you give me some tips ? 
Another question, why should I config all devices to the same MAC address in monitor as you refer it in the wiki?

xieyaxiongfly, 2018-01-03:
Ok, I met that before. I think it is caused by the ath9k dirver together
with the send_msg socket. I managed to make it work. But I have release the
code yet.

In OpenWRT, the lorcon works without problem. I will add Lorcon Support for
Ubuntu later.

kangqf, 2018-01-04:
Thanks a lot, I think this problem is cause by the rate control algorithm used by the kernel, May be the rate can't be set by the user. And one way to solve this problem that used by ` intel 5300 tool` is to modify the mac80211 and the driver as you refer to [here](https://github.com/dhalperi/linux-80211n-csitool-supplementary/issues/12). If you have other ideals to solve the problem please tell me, thanks a lot.

xieyaxiongfly, 2018-01-04:
Yes. It is not caused by rate control algorithm. It is a rate control
problem. Generally, it is because of the socket lorcon use to transmit the
packet doesn't respond to the rate set by the user. Anyway, we can change
it by modify the mac80211 or ath9k. I already solved it but I didn't do it
on Ubuntu. I did it on OpenWRT.

xieyaxiongfly, 2018-01-04:
Not OpenWRT, it's Linino, a special version of OpenWRT.

kangqf, 2018-01-04:
ok, thanks a lot for your explanation, I will refer your Linino version code then.
And, I want to give you some suggestion for the ubuntu version CSI tool, you may change the ath9k driver rather than the mac80211 module, because some users may only  rebuild the ath9k driver and replace the driver, they don't want to do any change for the kernel and kernel modules.  That's also the reason that I fork the repo ath9k from libing64 and add the [scripts](https://github.com/kangqf/ath9k/blob/master/install.sh) to rebuild ath9k.

DK-P, 2018-04-18:
Is the Injection patch for Ubuntu ready yet/ released anywhere?, If not, is there any guide/advice on modifications required to build your OpenWRT image for x86 hardware?
Thanks for your time.

junstriker, 2019-04-04:
Hello,
How to transmit packets of fixed length continuously from  AP (running atheros csi openwrt) to a PC (running atheros csi ubuntu) without having to type "sendData wlan0 xx:xx:xx:xx:xx:xx" every time at AP side?
Also, sometimes when I associate the PC to the AP and type: "sudo ./recv_csi log.dat", I start to receive packets of different lengths without even issuing the sending command at the AP. Is this normal?

Lastly, if I'm to record csi data continuously for an experiment, let say for 1 hour, how do I use the timestamp or any other parameters in the logged csi data to get a graph of csi amplitude versus time for the experiment duration? Does the timestamp include information about the date/time for each packet?
I would be really grateful if somebody can enlighten me on these.

Many thanks