xieyaxiongfly/Atheros-CSI-Tool-UserSpace-APP #41: Processing CSI with matlab

Furyt4nk, 2018-01-09:
Hi, I had a trouble when I accessed the measured CSI with matlab.

I tried to extract CSI value from the measured CSI binary file, but I failed.

It seems kind of formatting problem.

[image]

It seems work decoding the first packet, but It seems fail to decode from the second packet.

I tried to re-build mex file and run matlab in different environment, but anything worked.

Here I attach a screenshot.

Is there anything I did wrong or missed?

justinklchan, 2018-03-04:
I ran into this same issue and solved it.

It seems like the latest version of the file "read_log_file.m" is broken...

If you revert to a previous version of the file
https://github.com/xieyaxiongfly/Atheros-CSI-Tool-UserSpace-APP/commit/dbd3efa3c3ae8d56ddaa1f8cb5f1ac792a23ddc3#diff-21823fdb8f8d73dabdef736ad254426f

Then this error will go away and it will work.

The simple fix is on line 46:

endian_format = 'ieee-be';
should be 
endian_format = 'ieee-le';

brunosoaresds, 2018-09-26:
Maybe that can help you: https://github.com/brunosoaresds/csi-monitor

zhangdh24, 2019-01-11:
> I ran into this same issue and solved it.
> 
> It seems like the latest version of the file "read_log_file.m" is broken...
> 
> If you revert to a previous version of the file
> [dbd3efa#diff-21823fdb8f8d73dabdef736ad254426f](https://github.com/xieyaxiongfly/Atheros-CSI-Tool-UserSpace-APP/commit/dbd3efa3c3ae8d56ddaa1f8cb5f1ac792a23ddc3#diff-21823fdb8f8d73dabdef736ad254426f)
> 
> Then this error will go away and it will work.
> 
> The simple fix is on line 46:
> 
> endian_format = 'ieee-be';
> should be
> endian_format = 'ieee-le';

Thanks a lot! I am dealing with the same problem, It does work.

shulinyang, 2019-01-21:
Another issue is how to change the frequency to 5G.

army-mehak, 2019-04-19:
@brunosoaresds , how can I use your tool for plotting graph for offline ".dat" files?

muhmaz3, 2019-07-13:
I have tried the solution proposed by @zhangdh24. However, this did not work for me. I am using Windows 10 and Matlab 2018a. Here is how I fixed the read_csi.c file and read_log_file.m

In read_csi.c change the following lines, or basically, cast pointers to unsigned int pointers. Then compile the read_csi.c file by typing `mex read_csi.c` in command window. 

` nr_p = (unsigned int *) mxGetPr(prhs[1]); `
`nc_p = (unsigned int *) mxGetPr(prhs[2]);`
`num_tones_p = (unsigned int *) mxGetPr(prhs[3]);`

Also, you need to revert back to the previous version of read_log_file.m. I tried changing line 46 only as described by @zhangdh24 that did not work. A solution that worked for me is removing line 46 and adding the following from the [previous version](https://github.com/xieyaxiongfly/Atheros-CSI-Tool-UserSpace-APP/commit/dbd3efa3c3ae8d56ddaa1f8cb5f1ac792a23ddc3#diff-21823fdb8f8d73dabdef736ad254426f). 

```
endian_code = fread(f,1,'*ubit8');	
if endian_code == 255	
    endian_format = 'ieee-be';	
elseif endian_code == 0	
    endian_format = 'ieee-le';	
else	
    error('Wrong endian format.');	
end
```

Parbin, 2020-04-28:
@zhangdh24 It is not working for me ,altrough its completely running but i am getting the csi_len=0 for all ..
and as @muhmaz3 when i try its gives me an error "Wrong endian format"??
Can anyone help?? How should read the CSI matrix?

muhmaz3, 2020-04-28:
@Parbin 
It has been quite long, but the following solution worked for us. 
In **read_log_file.m** remove line 46 and add the following lines that are taken from the previous version.

```endian_code = fread(f,1,'*ubit8');	
if endian_code == 255	
    endian_format = 'ieee-be';	
elseif endian_code == 0	
    endian_format = 'ieee-le';	
else	
    error('Wrong endian format.');	
end
```