xieyaxiongfly/Atheros-CSI-Tool-UserSpace-APP #45: " Failed to open the device...: No such file or directory "

chinmoy009, 2018-04-18:
My laptop has Atheros AR9565 . I have installed linux kernel according to the Atheros-CSI-Tool-User-Guide. And after checking my kernel version i get the following
4.4.0-66-generic

Now, i am not actually sure about whether my installation is successful or not because when i am running main.c for receiving CSI , it says that  " Failed to open the device...: No such file or directory" . That means it is not finding "/dev/CSI_dev". 

It would be helpful if you can tell where is the problem and what is the solution. 

Thank you.

Dxploto, 2018-05-23:
Hey,chinmoy009. Have you solved this problem? I also try to install Atheros-CSI-Tool in Ubuntu 18.04 with kernel 4.2.10. And my Network Controller is Qualcomm Atheros QCA6174 802.11ac Wireless Network Adapter (rev 32).
I meet the same problem with you, which is not finding "/dev/CSI_dev".
I am not sure whether the problem is that my Wireless Network Adapter dosen't support Atheros-CSI-Tool nor I didn't install Atheros-CSI-Tool correctly.
I really appreciate it if you can help me , thank you.

yj1127, 2018-05-30:
Hi guys. I meet the problem too. Have you solved it?

Dxploto, 2018-05-30:
I tried to create an empty file "CSI_dev". But it seems makes no sense.

yj1127, 2018-05-30:
I guess so. My Network Controller is same with you. Are you create "CSI_dev" in recvCSI ?

Dxploto, 2018-05-30:
I create it in "/dev/CSI_dev". The recvCSI runs, but it seems dose not work correctly.

yj1127, 2018-05-30:
could you tell me about file path minutely? 😂 Are you create it in "Atheros-CSI-Tool/dev" or "Atheros-CSI-Tool-UserSpace-APP" ?

yj1127, 2018-06-03:
To Dxploto.
I solved the problem. 😎
I booted Ubuntu 4.1.10+ kernel version.
I use grub-customizer program.

xieyaxiongfly, 2018-06-11:
Hi there,

Your problem is because of the wrong system kernel. Your kernel is 4.4.0
which is not the default one. Generally, when you have installed a higher
kernel, 4.4 for example, the system will automatically select the higher
kernel.

When you run your 4.4 kernel, our tool is not installed.

xieyaxiongfly, 2018-06-11:
When you run kernel 4.2, our tool is not installed successfully. You can
select the kernel when the system starts.

Also, our tool doesn't support 11ac Wi-Fi cards. And my coding are based on
OpenWRT. I strongly recommend you to use our OpenWRT version.

chinmoy009, 2018-06-13:
Will it work on a 32 bit OS?? Did any of you install it on 32 bit OS?

To
yj1127

How can i downgrade my kernel ? It will be very helpful if you can provide
me with some proper resources.

xieyaxiongfly, 2018-06-13:
It is possible to select the Ubuntu kernel version when you start your PC.
Can you try that first?

chinmoy009, 2018-06-13:
I don't have kernel 4.1.10 in my grub menu.

xieyaxiongfly, 2018-06-13:
That's odd. After you successfully installed the tool, you should find
4.1.10 in the menu.

Here are several pieces of advice.
1, First of all, I strongly recommend you using our OpenWRT version. Check
this link <http://wands.sg/AtherosCSI/Hardware.html> to check what kind of
hardware to buy when you want to use OpenWRT version. I recommend WPJ558 or
TP-Link 4300. I am currently working on OpenWRT version and a lot of new
functionality are only supported in OpenWRT version. A few updates based on
openWRT is under preparing. Installing openWRT on those devices is easy and
there are huge amounts of tutorial about it.

2, If you still want to use Ubuntu version, you can install Ubuntu 14.04
and reinstall our tool on it.

Yaxiong

chinmoy009, 2018-06-13:
I will reinstall it . But the problem is I have 32 bit OS and in your documentation you recommend 64 bit OS . Moreover while running the command " make menuconfig " it suggests to choose 64 bit kernel ( as showed in attached image) . Is it possible to install on 32 bit OS ? 
[image]

xieyaxiongfly, 2018-06-27:
I think it won't work with 32-bit. Sorry, I forget to answer your question.

sayaz, 2018-08-13:
Hi,

I am using Atheros9462 NIC on Ubuntu 14.04 (64-bit)

After installation, I can see the update kernel version 4.1.10+ but I had some "depmod warnings" on the network interface.

I could not find any CSI_dev inside /dev. Also when I run ./recvCSI filename it gives "Failed to open the device...: No such file or directory".

Can you tell me what might be the possible reason behind this?

Parbin, 2020-03-01:
Hi all,

I am having the same problem ,I cant find CSI_dev under /dev folder.I am using 2 ubuntu (64- bit ) for transmitter and receiver ,In both the case i am not getting any CSI_dev ,although the kernel version is 4.1.10+ in both machines.
Could anyone please help me out??