xieyaxiongfly/Atheros-CSI-Tool-UserSpace-APP #47: How to convert amplitude  to amplitude in dB level

leewez, 2018-06-25:
Hi @xieyaxiongfly   I saw 2 pics in your website http://wands.sg/AtherosCSI/ showing amplitude and amplitude in dB level of four CSI.  Would you please share me some information about the connection between amplitude  to amplitude in dB level? how to convert amplitude  to amplitude in dB level ?

Thanks, 
Li, Yongfeng

xieyaxiongfly, 2018-06-25:
In matlab, you can use the db() function.

leewez, 2018-06-25:
@xieyaxiongfly  do you mean this function is provided by matlab? I'm wandering whether I can do this convertion in C language

xieyaxiongfly, 2018-06-25:
If you want to implement this function in c, you can take a look at how
db is calculated in matlab
<https://www.mathworks.com/help/signal/ref/db.html>and then do the same
operation in c. The db calculation is quite simple.