xieyaxiongfly/Atheros-CSI-Tool-UserSpace-APP #49: assignment from incompatible pointer type [-Wincompatible-pointer-types] using read csi mex file

aligharari, 2018-09-27:
Hi, I am trying to use your tool to receive CSI but I have some problems using your MATLAB code. The problem is that when I try to make mex file I face the following warnings. After running the code I face the error which tells MATLAB faced an internal error and close the MATLAB. I will appreciate it if you can help me with my problem.
read_csi.c:67:11: warning: assignment from incompatible pointer type [-Wincompatible-pointer-types]
     nr_p  = mxGetPr(prhs[1]);
read_csi.c:70:11: warning: assignment from incompatible pointer type [-Wincompatible-pointer-types]
     nc_p  = mxGetPr(prhs[2]);
warning: assignment from incompatible pointer type [-Wincompatible-pointer-types]
     num_tones_p = mxGetPr(prhs[3]);

kdkalvik, 2018-12-17:
try changing the endian_format in read_log_file.m

muhmaz3, 2019-07-12:
Did you find a solution to this, I am also stuck at this point.

muhmaz3, 2019-07-14:
This will help.
https://github.com/xieyaxiongfly/Atheros-CSI-Tool-UserSpace-APP/issues/41#issuecomment-511161465