xieyaxiongfly/Atheros-CSI-Tool-UserSpace-APP #51: Broadcast mode in ubuntu and ways to work in 5G

jinzhang-unsw, 2018-12-24:
Does the ubuntu version work in broadcast mode?
I follow the procedure and use the script:
'''
service network-manager stop
ifconfig wlan0 down
ifconfig wlan0 hw ether 01:F0:A1:98:42:41
ifconfig wlan0 up
service network-manager start
iwconfigure wlan0 essid Atheros_csi_tool
'''
The previous script does change hw address but the receiver still cannot get CSI data.

Another issue is how to change the frequency to 5G. Any necessary changes are needed in hostapd configure file?

Many thanks.

shulinyang, 2019-01-22:
Can you solve this problem?

shulinyang, 2019-01-22:
how to change the frequency to 5G

shulinyang, 2019-02-27:
ubuntu模式下如何修改信道呢

shulinyang, 2019-04-11:
how to change the frequency to 5G@jinzhang-unsw