xieyaxiongfly/Atheros-CSI-Tool-UserSpace-APP #58: How to plot CSI amplitude vs Time

abhinav2938, 2019-03-28:
Hi,
I am able to receive the CSI amplitude vs no. of packets graph. But I don't know how to plot it in time domain. I know that the graph of CSI amplitude vs no. of packets gives data in time domain, but I want it specifically to plot amplitude vs time because I need to further process it for generating spectrogram. I am attaching one screenshot of my plot.
Any help is highly appreciated.
[image]

shulinyang, 2019-04-15:
请问您是用路由器进行发送吗

army-mehak, 2019-04-19:
Hi @abhinav2938
I wish I could help you!
I have a question. How did you manage to calculate amplitude vs packet number?
From @xieyaxiongfly tool, I have managed to read my ".dat" file on Matlab.
[image]

How did you calculate the amplitude?
Thank you!!

abhinav2938, 2019-04-20:
Hi,

Calculating the CSI amplitude is very easy once you get the CSI complex data. Just take the absolute value of the CSI amplitude and use _db_ to convert it into the decibel scale. For plotting the CSI amplitude vs no. of packet just create a loop which runs n times(no. of packets collected). e.g. if you collected 100 packets, just create a loop which runs 100 times and calculates the value of CSI amplitude for each loop. Then you can plot the CSI amplitude.

army-mehak, 2019-04-20:
As what I understood, I did this in matlab
[image]

Am I right?
How would I plot then?

abhinav2938, 2019-04-20:
Did you try looking at this page:
[get_csi_streams](https://github.com/brunosoaresds/csi-monitor/blob/master/csi-monitor-gui/get_csi_streams.m) it is really helpful and easily available. 
First, you need to streamline the CSI data from each Tx-Rx pair, i.e. collect the CSI data from one pair(in your case it would be 4 as 2*2). Once you store it in a cell array, you can use proper indexing to get the CSI value from one pair, in a matrix form and then use plot function to plot the graph.

The above code converts the CSI complex values into db values and again store it in a cell array, you can't plot from a cell array. you will have to extract the CSI amplitude value in a matrix form and then you can plot function to plot it.

vibhorrawal, 2020-02-28:
Still not able to plot, after implementing above solution

SumaiyaRahmat, 2022-02-18:
can anyone please tell me how to plot csi amplitude vs no of packets in matlab?