xieyaxiongfly/Atheros-CSI-Tool-UserSpace-APP #66: Irregular reception of CSI packets

muhmaz3, 2019-07-12:
I have two TP-WDR4300 routers, on both routers, I flashed the Openwrt with Atheros-CSI-Tool. For the reception of CSI packets, I used AP mode as suggested in the guide.  Then I run recvCSI on the client and send packets from the AP using sendData. However, when I send 10 packets, I receive 5 or six packets, sometimes I get only 1,2, or 3 packets and sometimes I get nothing. I am not able to figure out what is going wrong in my case, why I am not able to receive all sent packets.  

Secondly, I wanted to ask is there a way to have fix data rate for every packet? and how I can configure to use only 1 antenna for transmission.

abid1403, 2019-09-23:
I have the same problem. Do you have any update how to solve this problem

muhmaz3, 2019-09-23:
This problem happens due to data rates. Refer to the following and I hope that will solve the issue you are facing. 

https://github.com/xieyaxiongfly/Atheros-CSI-Tool-UserSpace-APP/issues/22#issuecomment-512192196

May I know which mode you are using, injector, access point, or broadcast?

abid1403, 2019-09-23:
Thank you for your helpful answer. 

I use AP mode.

magienguyen, 2019-09-26:
@muhmaz3 Currently I am using a TL WDR 4300 v1.7 and a PC and csi_len is always 0. I follow your way to check the data rates and this is the results 
root@OpenWrt:~# iw phy0 info | grep "MCS rate indexes"
		HT TX/RX MCS rate indexes supported: 0-15
root@OpenWrt:~# iw phy1 info | grep "MCS rate indexes"
		HT TX/RX MCS rate indexes supported: 0-23
I checked the rate table but there is no rate like mine. What is your rate? We are using the same device so I want to check it, or could you show me the way to fix it? thanks

muhmaz3, 2019-09-26:
@magienguyen 
I did not understand the following statement of yours. Are you referring to the rate table given in the PDF?
> I checked the rate table but there is no rate like mine. 
Also which mode you are using AP or monitor? My devices are currently configured in the monitor mode and I use 0 indexed data rate which is 128 (decimel) or 0x80(hex).

magienguyen, 2019-09-26:
I use AP mode, where TP-link 4300 version 1.7 is AP and a PC as client. I flashed the router with the provided firmware on Github. When I use send_Data (from the OpenWRT User Space) on my PC and recv_CSI on the router, the rate is pretty normal I think (0x80-0x8b). However, the CSI is always 0. I think the monitor mode is basically the same with AP mode, the difference is just about the number of clients. Can you think of any reason? Thanks

magienguyen, 2019-09-26:
I even configure the 4300 to be a dumb AP, which means I connected the router to an existing Netgear router which has internet connection. I turn off DHCP on the 4300, give it another IP (different from the Netgear). I just use recvCSI on the 4300, watch some videos from Youtube, use Facebook and let it collect data from that. The CSI I got from that data is also 0

magienguyen, 2019-09-26:
Also, how did you set up monitor mode? Did you use two computers to ssh to two routers?