xieyaxiongfly/Atheros-CSI-Tool-UserSpace-APP #67: I have one question about receive csi.

jiandan521, 2019-07-14:
I tried using a desktop computer with the atheros AR9380 network card installed to send data to the router and receive csi. (the router was not processed at all).But I didn't get csi. What's wrong?

[image]

muhmaz3, 2019-07-16:
What do you mean by 

> the router was not processed at all

 If I understood your question correctly, I think on both devices you need to install the CSI tool.

CSIhicham, 2022-02-23:
Hi,

I have the same problem with the TL-WR1043ND v1 router and the Qualcomm Atheros AR5BWB222 Wi-Fi network card on PC?

Have you solved this problem?

Thanks,

CSIhicham, 2022-07-04:
Hi Itamar,

T think that TL-WR1043ND v1 not support Ahteros CSI Tool.